alcar21/WemosEM #1: platformio compile problems

dpavlin, 2019-10-05:
First, thanks for such a nice project. I'm having problems compiling using PlatformIO, version 4.0.3 (which is currently latest). When using (default) platform_150 it doesn't fit:

```
/home/dpavlin/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini/firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pio/build/d1_mini/firmware.elf] Error 1
```

And when compiling with all others I get:

```
Setup finished

Exception (3):
epc1=0x4021c03c epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025c5d7 depc=0x00000000
```
And esp keeps rebooting.

Your release binary does work for me, so I'm wondering how did you compile it?

alcar21, 2019-10-07:
Please, compile with platform_180. I fix as soon as possible.
In the current download of the arduino versions two files are missing. You can download them from here: https://github.com/esp8266/Arduino/tree/2.3.0/tools/sdk/ld
The files are: eagle.flash.4m3m4s.ld and eagle.app.v6.common.ld
I will correct it later.

alcar21, 2019-10-07:
Sorry, fixed in master branch.

closed 2019-10-07 by alcar21