alcar21/WemosEM #3: MQTT not enabled after reboot

mason-xx, 2019-11-23:
Flashed WemOS D1  mini with WemosEM-1.3.1-core_2.5.bin. Configured and enabled MQTT on the Webif. Working fine. But after disconnect the Wemos aka reboot it, MQTT stays disabled and need to be reenabled by the webif to work again.

Expected behaviour would be to stay activated after a reset/reboot of the controller.

alcar21, 2019-11-23:
Please, use version WemosEM-1.3.1-core_2.4.bin. 
The arduino 2.5 core has problems with the Wifi controller. I have read it on the [Tasmota website](https://github.com/arendst/Tasmota/releases) **search** "Core version 2.3.0 vs 2.4.2 vs 2.5.2"

The core 2.4 in this firmware working fine. Please, can you attach the logs if you know how to obtain them?

> Expected behaviour would be to stay activated after a reset/reboot of the controller.
yes, this is how it works for me with WemosEM-1.3.1-core_2.4.bin. I honestly haven't tried the version with core 2.5.

Thank you.

mason-xx, 2019-11-23:
You want the logs of the 2,5 version or the 2.4?

alcar21, 2019-11-23:
The test performed by other developers, the version 2.4 is usually more stable. The best core is 2.3 but there are libraries that use firmware that does not support 2.3.
You can attach the logs of the version you want but I recommend you use the 2.4

mason-xx, 2019-11-23:
Here is a log from 2.5 https://hastebin.com/awoqifehiz.cs not much in there..
But after reflashing with 2.4 mqtt stays enabled after a reboot as expected... thank you! :)

alcar21, 2019-11-24:
Great!
Log is only necessary when the problem occurs but it is fully proven to fail.
Thank you very much.

closed 2019-11-24 by alcar21