alcar21/WemosEM #4: Domoticz integration

InvertedMinor, 2019-11-27:
Could you provide a Domoticz integration via MQTT?

alcar21, 2019-11-28:
Ok, I Try. I have not Domoticz .

InvertedMinor, 2019-11-28:
Great:  https://www.domoticz.com/wiki/MQTT

alcar21, 2019-11-28:
it is easy. Please, Can you do beta tester??
Thanks!

InvertedMinor, 2019-11-29:
of course!

asem67, 2019-12-31:
Could you provide a Domoticz integration via MQTT 
or IoT Connections >> [Domoticz API](https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Electricity_.28instant_and_counter.29) 

Thanks !

alcar21, 2019-12-31:
Hello,
I'm working in it. in the next few days I will release a version compatible
with domoticz
Thanks

pepijnkoopman, 2020-01-12:
Looking forward to the Domoticz integration via MQTT! When it's is ready I would like to beta test as well.

RichieRogers, 2020-05-11:
Hi,
It looks like development has stopped on this :(
Has anyone successfully integrated this with Domoticz (ideally with MQTT)?
If so, can you supply the firmware or what modifications i need to make to WemosEM to get this working?

Thanks,
Richie

pepijnkoopman, 2020-05-11:
Hi Richie, I integrated WemosEM with Domoticz using **MQTT + Node-RED**. WemosEM publishes the data on a MQTT server (Mosquitto). A Node-RED script subscribes to these messages on Mosquitto server, converts them to messages for Domoticz and publishes the messages to the domoticz/in topic of the MQTT server of Domoticz. More information: https://www.domoticz.com/wiki/MQTT

RichieRogers, 2020-05-11:
Ahh, ok.
I'm running Domoticz and Mosquitto on a Windows environment, so will look at installing Node-Red as well (no experience of Node-Red so will be interesting getting this working).

Thanks,
Richie

pepijnkoopman, 2021-05-26:
Hi @daniel-dms, below is the export of my Node-Red flow:
`[{"id":"3f2dd754.b751f","type":"tab","label":"PowerMon1 Current","disabled":false,"info":""},{"id":"40a67cbc.ee3d74","type":"mqtt in","z":"3f2dd754.b751f","name":"","topic":"wemos/wemosEM-239E74/power","qos":"0","datatype":"auto","broker":"bbc5f10.1ab581","x":170,"y":60,"wires":[["da04248a.acb1a"]]},{"id":"da04248a.acb1a","type":"json","z":"3f2dd754.b751f","name":"","property":"payload","action":"","pretty":false,"x":350,"y":220,"wires":[["a14cc7d1.c7e93"]]},{"id":"a14cc7d1.c7e93","type":"function","z":"3f2dd754.b751f","name":"Get current","func":"var value = msg.payload.current;\nmsg.payload = value;\nreturn msg;","outputs":1,"noerr":0,"x":530,"y":220,"wires":[["9769d2ff.7a808","f3521a2e.f786e8"]]},{"id":"709c2826.f4fb28","type":"mqtt out","z":"3f2dd754.b751f","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"72fca658.733588","x":1030,"y":60,"wires":[]},{"id":"e750ff81.9554d","type":"debug","z":"3f2dd754.b751f","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":990,"y":300,"wires":[]},{"id":"9769d2ff.7a808","type":"debug","z":"3f2dd754.b751f","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":690,"y":300,"wires":[]},{"id":"f3521a2e.f786e8","type":"function","z":"3f2dd754.b751f","name":"Add Domoticz IDX 33 (PowerMon1Current)","func":"var current = msg.payload;\n\nmsg.payload = {};\nmsg.payload.svalue = current;\nmsg.payload.idx = 33;\nreturn msg;","outputs":1,"noerr":0,"x":870,"y":220,"wires":[["e750ff81.9554d","709c2826.f4fb28"]]},{"id":"bbc5f10.1ab581","type":"mqtt-broker","name":"SDRPi","broker":"192.168.2.31","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"72fca658.733588","type":"mqtt-broker","name":"","broker":"192.168.2.48","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"15","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]`

daniel-dms, 2021-05-26:
THANKS.........IT'S WORKING