alcar21/WemosEM #6: A couple of minor items and a question

NorthernMan54, 2020-01-20:
Tks for this piece of work, it is very well done.  I just have a couple of minor things

1 - I'm in North America and I can't set the timezone to -5 ( Eastern Standard )

2 - When the device reboots, the data is wonky for the first minute, can this false data be thrown away?  ie call em_read(true); at the end of the setup function?

3 - When Power usage is zero, the current sits between 0.02 to 0.04, any thoughts on why?

For setup, I'm using a SCT013 Calibrated at 30A 1V, local power is 120 volt 60 Hertzs

alcar21, 2020-01-22:
Hello, thanks.

1. Ok, i will fix in next version.
2. It's difficult, when system start, it perform a calibration and discard initials values. I could add an option in configuration that discard data of a period of time
3. It is the rounding of precision, it will really count for example 0.004 watts
4. Which version are you using? Please use the core 2.4

The measured is correctly? Thanks by the report.
Please be patient, my dedication time to the project is very short.

NorthernMan54, 2020-01-22:
For these, I'm putting together a pull request and fix for the MQTT Issue

I'm just testing behaviour when WiFi drops and restarts, then I will submit

alcar21, 2020-01-22:
Thank you.

Un saludo,
Alfonso C.

NorthernMan54, 2020-01-23:
Please review pull request #8

alcar21, 2020-01-23:
Accepted. Thanks.

NorthernMan54, 2020-01-23:
Tks

closed 2020-01-23 by NorthernMan54