alcar21/WemosEM #7: MQTT gets disabled after a reboot

amoros1, 2020-01-22:
Congrats for the nice piece of work!

I found that after enabling MQTT and saving in the MQTT parameters, when the Wemos restarts, the MQTT gets disabled so you need to enter again in the UI to enable it. I wonder if there is some issue with the enable flag that after a reboot it is set again to disable?. 
Could be this behavoir the way this function was designed to work?, 

Many thanks
Luis

alcar21, 2020-01-22:
:smirk: yes, this error has been reported by other people in private. I'm working in it.

if yoy want try, in the previous version if it works correctly, I am checking it.

Thanks, regards.

amoros1, 2020-01-24:
Thanks. Yes, I will make the test
Regards

plutec, 2020-04-06:
Are working in the previous version? Did you fix it in the new one?

Thanks!

alcar21, 2020-04-06:
Sorry, I have yet to release the patch. This week i try it.

PaulHam211, 2020-04-21:
Hey, Any chance of releasing the patch for this? I have set up my device today and have noticed that MQTT disables after a reset.
I considered trying to build a version with MQTT enabled in code however I struggled to build in Atom (mac equivalent of Platform IO) and now I assume this is a bug and enabling MQTT automatically wont actually fix it?

nandosof, 2020-05-14:
have fix?? thank you!

alcar21, 2020-06-02:
Sorry, I will release a version this weekend.

I don't have much time to dedicate to the project.

Thanks you

PaulHam211, 2020-06-21:
Sorry @alcar21  I know your busy but any luck with this MQTT enable on boot fix?

plutec, 2020-07-06:
Fixed in code, I just compile for Core 2.4. I didn't check yet in my hardware.

https://mega.nz/file/eRknmQKQ#T9rFCZ6qjTx1qB8fpxCNPB3yId9L3Dg5LpY-535Yt5k

plutec, 2020-07-06:
I just checked in my wemos and at least it saves the MQTT after restart. So this flaw can be considered as solved.

nandosof, 2020-07-06:
Work fine. Thanks for your work!

rechena, 2020-10-04:
I can confirm that on the latest version this issue still happens :(

plutec, 2020-11-01:
@rechena is your issue related with MQTT or with WiFi connection? Please, take a look to this PR: https://github.com/alcar21/WemosEM/pull/16

rechena, 2020-11-01:
> @rechena is your issue related with MQTT or with WiFi connection? Please, take a look to this PR: #16

My issue is with MQTT, when I restart the device MQTT goes back to disabled :(

plutec, 2020-11-01:
Did you try with the version I shared un this issue?

plutec, 2020-11-02:
@rechena please, update with this version: https://mega.nz/file/eRknmQKQ#T9rFCZ6qjTx1qB8fpxCNPB3yId9L3Dg5LpY-535Yt5k
And then, configure the MQTT again. For me it's working perfectly.

rechena, 2020-11-03:
> @rechena please, update with this version: https://mega.nz/file/eRknmQKQ#T9rFCZ6qjTx1qB8fpxCNPB3yId9L3Dg5LpY-535Yt5k
> 
> And then, configure the MQTT again. For me it's working perfectly.

Oh will give it a try. Can I use the OTA feature? Or do I need to plug cable? Thanks

plutec, 2020-11-03:
With OTA is ok

rechena, 2020-11-03:
Yay, just updated using OTA and its working just fine, didn't even have to reconfigure MQTT :) Thanks

plutec, 2020-11-03:
@alcar21 you can close this issue. Thanks!

alcar21, 2020-11-08:
Thank you. Fix in version 1.3.2

closed 2020-11-08 by alcar21