alcar21/WemosEM #11: Disconnects...

mason-xx, 2020-05-28:
Hey there...

using wemosEM for month now, was fine the first few. Right now the D1s disconnect random every hour/day and I have to reset them manually.  Someone has an idea how to debug this?

Every input is welcome, I'm a little lost here...

catatonico2, 2020-06-02:
I have the same problem, I lose the connection and to be able to reconnect I have to restart.

alcar21, 2020-06-02:
Hello,

Sorry, I have yet to upload a correction that solves the problem but for
work and family reasons I have not been able to upload the binary

Thanks.

mason-xx, 2020-06-03:
every piece is appreciated, thank you! :)
but not sure if its a dupe since in that case the d1 isn't reachable via wifi too and not only mqtt.

sagatxxx, 2020-06-19:
thanks for your work alcar, when you can...if you could fix that when the wifi is missing and It come back again can leave the AP mode it would be great.  For example, when in AP Mode, scan the Wi-Fi in case it is available again and connect again.

catatonico2, 2020-08-07:
Alca21, solo por saberlo si se va a poder solucionar las desconexiones. Si no buscar alternativas. Gracias por tu trabajo.

alcar21, 2020-08-07:
Hello,
Yo have a solution in : https://github.com/alcar21/WemosEM/issues/7#issuecomment-654202414
Regards.

plutec, 2020-11-03:
@mason-xx @catatonico2  I think this issue is fixed in the master branch. Please, compile and check it. Thanks!

alcar21, 2020-11-08:
Fixed in release 1.3.2

closed 2020-11-08 by alcar21