alcar21/WemosEM #12: 1v vs 3.3v

buzzy, 2020-07-01:
How did you solve the fact that the CT outputs 0-1v and the Wemos D1 has a built-in voltage divider on the ADC that allows 0-3.3v on the ADC? If you don't boost the CT output, you will only use 1/3 of the resolution of the ADC, giving you only 333 steps between 0A and 30A, about 0.1A steps, which would mean 22W "resolution". Please advice.

rechena, 2020-10-04:
Would this cause the power consumption to be reported way lower than expected? For e.g a 2500W being reported as ~500W for example? Thanks

alcar21, 2020-10-05:
The ADC of ESP8266 is 0-1v please read the documentation:
https://arduino-esp8266.readthedocs.io/en/latest/reference.html

A good improvement will be using the ADC of ADS115 that it has 0-5v. I am
working on a version that it's better and it uses the module PZEM-004T.

Un saludo,
Alfonso C.

closed 2020-11-08 by alcar21