alcar21/WemosEM #14: Monitor the Mains?

rechena, 2020-09-20:
Can I use this board code to monitor my house at the mains board? Thanks

alcar21, 2020-09-20:
yes of course.

Un saludo,
Alfonso C.

rechena, 2020-09-21:
Thanks for the reply. I had a old system using an arruino uno and Xbee but this makes it so much better :) 
One thing though on mine I was measuring using an AC/AC transformer for the current, would it be possible to do the same somehow on yours? Thanks

alcar21, 2020-10-01:
Yes, you can measure AC by putting the clamp on one phase. You only need a transformer to power the ESP8266

rechena, 2020-10-02:
Thanks I was more asking about using this approach - https://learn.openenergymonitor.org/electricity-monitoring/ctac/how-to-build-an-arduino-energy-monitor. But That would need some changes on the board. 
Regardless, do you have any link to help with the calculation of the icalc? Thanks

alcar21, 2020-10-02:
Yes, please, visit this url:
http://lechacal.com/wiki/index.php/Current_Transformer_Conditioner

Un saludo,
Alfonso C.

rechena, 2020-10-04:
Hi, I'm a bit confused on whats happening, I finally got around to install this, ufortunately I had to install the CT on the Neutral since the Live is too short, regardless, from what I read it won't make a difference. 
The concern is, looking at the house in "standby" I'm reading around 146W, which I love it being so low, but.. when I turn on my kettle which is 2500/3000W, the Watts only increase to 568W... I have a SCT-013-030 and I've calibrated the iCalc to 30.0... I've measured my Voltage with a Multimeter and its giving me 233V.

Can you think if anything I might be doing wrong?

Thanks

alcar21, 2020-10-05:
yes, all ok. it's strange.

Please, Can you probe a toaster or a hair dryer?

Thank you.

rechena, 2020-10-05:
I would gladly but unfortunately it was so hard to had the CT to the main's board that if I remove it it will break :( need to order a new one and then I can test. I was just wondering if you noticed something similar... Guess I need to buy a new CT and test it... :(

alcar21, 2020-10-05:
it's not necessary. You connect the hairdryer o the toaster in a socket and
test.

rechena, 2020-10-05:
I've just tested both ways, with the socket and directly with just the hairdryer, and its reporting only ~400W for a 1800-2200W hairdryer :( 

[image]

alcar21, 2020-10-05:
Please, Can you post images of your configuration?

rechena, 2020-10-05:
Of the d1mini and all the components?

alcar21, 2020-10-05:
Only configuration calibration tab of wemos d1 mini. this:

[image: image.png]

alcar21, 2020-10-05:
[image]

rechena, 2020-10-05:
Sure :) 
[image]

btw, I've measured the Voltage with a multimeter on a wall plug.

alcar21, 2020-10-05:
oook, CT is 0-1v or 0-5v?

rechena, 2020-10-05:
It's the SCT-013-030...so from the specs its 0-1v? I did used in the past with a 5V Arduino, not sure if it helps.

alcar21, 2020-10-05:
Arduino works with 5v and ADC of ESP8266 work with 1v. This board is calibrated for the ADC of the Wemos D1 to range 0-1v.

[image]

rechena, 2020-10-05:
That is the board I got, I got it from the eBay link you have on this page.. :( and I'm using it with a Wemos D1

alcar21, 2020-10-05:
It's all Ok. Can you post a photo of the connections with Wemos D1, board, and CT?

rechena, 2020-10-05:
[image]
[image]
[image]

rechena, 2020-10-05:
I might be wrong, but I find the A/Current value way to low...

alcar21, 2020-10-05:
yes is very low. Sure, this is not well.

I believe that it's the cable of CT (jack 3.5) that connects to the board.
I have a friend that has the same problem but I am not sure.

rechena, 2020-10-05:
So you think I might have a bad CT? :(

alcar21, 2020-10-05:
yes, may be

rechena, 2020-10-05:
Ok so not much to do but buy a new one. Any recommendations on the Amp? Should I keep the 30amp or go higher?

alcar21, 2020-10-05:
I recommend to you buying PZEM-004T. I have a firmware that it's compatible with PZEM-004T.
This CT has more precession that SCT-013-30. The SCT-013 is best when it's smaller.

rechena, 2020-10-05:
Does it work with the d1mini and the shield? Do you have a link?

rechena, 2020-10-06:
Was just checking the eBay where o got the board and I think I've figure out the issue. This board is for use with the sct-013-000 so I'm guessing there's an extra burden resistor, I do recall when I was using the openenergy schematics that was the plan. I've just ordered one and will let you know.

alcar21, 2020-10-06:
Indeed, I had not noticed.

WARNING: If you use a calibrated clamp, remove resistance R1 from mains
current sensor.

Un saludo,
Alfonso C.

rechena, 2020-10-06:
Don't think mine is calibrated... and wouldn't want to remove  an R1.. so I might just wait for the 000 one to arrive :)

alcar21, 2020-10-06:
you remember that you must modify the ICAL value for STC-013-000. This
current transformer has minor precision

rechena, 2020-10-06:
Yep, will make the change which from this link: http://lechacal.com/wiki/index.php/Current_Transformer_Conditioner is 85.71... interesting number... :D

alcar21, 2020-10-06:
jejeje, perfect! I hope that it's will go fine.

Regards,
Alfonso C.

rechena, 2020-10-06:
After a chat with the board seller he confirmed that my CT already had a burden resistor, so I've just removed resistor 1 and now I have proper measures :)

Standby: [image]

Kettle: [image]

alcar21, 2020-10-06:
great!

rechena, 2020-10-06:
Thanks for all your help. I'm gonna close the issue :)

closed 2020-10-06 by rechena

sachasmart, 2025-02-02:
Very sorry to open this again, but attempting to get my house connected. I have this flashed to a Wemos D1 mini n1, which has the [Mottrams Lab - MLP191020](https://www.ebay.co.uk/sch/i.html?_nkw=&_armrs=1&_ipg=&_from=&_ssn=mlabs2018) hat. I am using a [100A/50mA CT clamp](https://www.aliexpress.com/item/32231711500.html?spm=a2g0o.order_list.order_list_main.5.2bff1802d1NOaR). 

I have this over a cord to my homelab, which should be pulling around ~400W. 

[image]

[image]

[image]