alcar21/WemosEM #15: Relay shield

avm162, 2020-10-29:
I first congratulate you on the work. It works very well to monitor consumption. It would lack a little more precision.
I wanted to know if it is possible to integrate at least one output to be able to put a relay shield. If it could be controlled by mqtt and even depending on some consumption rule it would be fantastic. Thanks anyway if it can't be done.
Greetings.
Sorry for the English, I used Google translator

plutec, 2020-11-01:
Hi @avm162 .
I think than in the hardware device you have enough GPIO to use, but the functionality of the relay is not really related with the purpose of the project. Maybe you can manage that with any other tool (check ESPHome). If your requirement is a bit more complex, you need to jump to HomeAssistant to create automations.

avm162, 2020-11-01:
Hi.  It was to control a relay depending on the consumption, even through
HA, but without having to use another esp.  thanks anyway.

plutec, 2020-11-01:
Yes, I understand what you say, but is not the purpose of the firmware. Just waiting to @alcar21 to confirm if wants to include that feature.

alcar21, 2020-11-08:
I understand good. No problem 👍 
The problem with precision is because the ESP8266 has an ADC that doesn't allow more precision to be measured.

I am preparing another firmware for PZEM 004T, it is much better.

it's possible to use a relay shield but it's better that you use another system like a Tasmota, Espurna, or EspHome to control this relay shield. Another option is to use one output to connect a contactor.

Now, I have no time. Due to the pandemic, it is difficult for me to spend time on this firmware.