alcar21/WemosEM #18: MQTT failure on Home assistant

Quercus84, 2020-11-06:
Antes de nada, muchas gracias por el proyecto Alfonso! Sigo en inglés.

I have a problem with the WemosEM publishing via MQTT on Home assistant. I've created the mqtt sensors and configured the server info on the WemosEM but the sensors have no data.

On the mosquitto broker log I have this message repeated several times:

1604686147: New connection from 192.168.1.172 on port 1883.
1604686147: New client connected from 192.168.1.172 as wemosEM-E3ED32 (p2, c1, k30, u'ender').
1604686193: Client wemosEM-E3ED32 has exceeded timeout, disconnecting.

Someone can send me in the right direction? Thanks

alcar21, 2020-11-08:
@Quercus84 do you have a telegram? I would like to see logs of wemosEM.

In MQTT, Can you see the topic of WemosEM?

Thank you.

Quercus84, 2020-11-13:
Problem solved! The wemos was in a different subnet and apparently that was the problem.

closed 2020-11-13 by Quercus84

alcar21, 2020-11-13:
Great!