alcar21/WemosEM #19: Do I remove the R1 resistor?

Quercus84, 2020-11-06:
I have a SCT-013-030 and my readings are off when using the ICAL proposed on http://lechacal.com/wiki/index.php/Current_Transformer_Conditioner. I need to push the ICAL up to 140 to get something reasonable. However, I was using the firmware from the domology tutorial and my readings were correct with the hardware as is.

I saw that @rechena ended up removing his in this issue #14 but I'm not sure if I should too. 

So, do I take it out? 

Thanks!

rechena, 2020-11-06:
If you have the same shield as I had yes you need to remove resistor 1 as it's a burden resistor which already exists in the 030. Alternative is to get a 000 ct.

Quercus84, 2020-11-06:
Ok. I'll try that. Hopefully nothing burns :p 

With what ICAL did you ended up?

rechena, 2020-11-07:
I ended up with 30ical, one suggestion on the resistor is remove only one
side, so if in the future you buy a 000 you could just solder the resistor
back.
I ended up buying another board and a 000 but mostly because my house often
goes above 30amp. I still get measures with this one but to be more
precise.

Hope it help

Quercus84, 2020-11-13:
I removed the resistor and the readings improved. However I am still calibrating the system. With an ical of 30 the readings are overestimated. I'll put the final number when I get it.

alcar21, 2020-11-13:
The measurements aren't precise because it has a margin of error, sometimes great.
I recommend that you calibrate the average power that you use at home. For example, if you consume an average of 2kw, perform the calibration with a device that measures more or less that.
The value of Ical 30 is approximate.

Quercus84, 2020-11-13:
I am calibrating using the daily total in kwh. Currently with ical=26 is still a bit high.

Seraf, 2022-08-26:
Hello,

Maybe someone here may help me 🙏🏻 
I bought the same 30A model, so I removed the R1 from the shield.
I don't think my current problem is because of this R1, as I tested different wemos and different shields (with and without R1):

* When I start the wemos without the shield, I can access the wifi, access the webserver.
* When I hotplug the shield, I can still access the wemos, but I can't get any relevant result. The led of the shield is off.
* When I boot the wemos with the shield attached, the led of the shield turns on (and the led of the wemos turns off), but I can't access at all the wifi. I have no connection to it.

I'm using the last 1.3.2 firmware version.
I tested with multiple powers, computer, phone charger etc ...
I verified with a multimeter, I have my 5V on the pins of my wemos, but I also have power on my shield etc ...
I don't know what I'm doing wrong ..
[image]
[image]