alcar21/WemosEM #20: Calibration how to

drybrdl, 2021-02-09:
Hi, 
  is it possible to add a small "how to" for the calibration to the readme file. The link in the app for the Current_Transformer_Conditioner calibration doesnt really make much sense as its not the same device.
Im sure a quick "how to" for working out your settings would be helpful to alot of people including myself

Thanks for the great work

alcar21, 2021-03-21:
Ok, when I have a bit little time, I update the readme with the calibration section.
The best value for the calibration is the average consumption of your house or devices that you monitor.

alcar21, 2021-04-03:
@wladimir79 you can view the ICAL value in http://lechacal.com/wiki/index.php/Current_Transformer_Conditioner

CT | Output Type | Rburden | Rgain (for 5V) | ICAL (for 5V) | Rgain (for 3.3V) | ICAL (for 3.3V)
-- | -- | -- | -- | -- | -- | --
SCT-013-005 | Voltage | None | 130 kΩ | 2.83 | None* | 5.0
SCT-013-010 | Voltage | None | 130 kΩ | 5.66 | None* | 10.0
SCT-013-015 | Voltage | None | 130 kΩ | 8.49 | None* | 15.0
SCT-013-020 | Voltage | None | 130 kΩ | 11.31 | None* | 20.0
SCT-013-025 | Voltage | None | 130 kΩ | 14.14 | None* | 25.0
SCT-013-030 | Voltage | None | 130 kΩ | 16.96 | None* | 30.0
SCT-013-050 | Voltage | None | 130 kΩ | 28.28 | None* | 50.0
SCT-013-060 | Voltage | None | 130 kΩ | 33.94 | None* | 60.0
  |   |   |   |  
SCT-006 | Current | 10 Ω | 17 kΩ | 11.62 | 28 kΩ | 17.5
SCT-013-000 | Current | 10 Ω | 40 kΩ | 56.57 | 75 kΩ | 85.71
SCT-019 | Current | 10 Ω | 24 kΩ | 116.13 | 40 kΩ | 171.43

wladimir79, 2021-04-05:
I saw this link, but I did not know whether to take the value for 3v or 5v sorry

alcar21, 2021-04-05:
My apologies, It's value of 5v.