alcar21/WemosEM #21: Mqtt Problem

wladimir79, 2021-03-21:
Hi,
i configured MQTT Parameters and from Mosquitto broker logs the device is connected.
```
1616335675: Client wemosEM-4E7EF6 already connected, closing old connection.
1616335675: New client connected from 192.168.1.54 as wemosEM-4E7EF6 (p2, c1, k30, u'mqtt_user').
```

The entity created on Hassio always returns UNKNOWN
this code is in sensors.yaml

```
- platform: mqtt
  state_topic: "wemos/wemosem-4e7ef6/power"
  name: "EM General A"
  icon: mdi:current-ac
  unit_of_measurement: "A"
  value_template: "{{ value_json.current }}"
- platform: mqtt
  state_topic: "wemos/wemosem-4e7ef6/power"
  name: "EM General W"
  icon: mdi:power-plug
  unit_of_measurement: "W"
  value_template: "{{ value_json.watios }}"
- platform: mqtt
  state_topic: "wemos/wemosem-4e7ef6/power"
  name: "EM General KWh"
  icon: mdi:power-plug
  unit_of_measurement: "KWh"
  value_template: "{{ value_json.kwh }}"
```

Where am I doing wrong?

thx

mason-xx, 2021-03-21:
why do you even need the yaml config? hassio is able to autodiscover mqtt devices and creates the entities for you?
Just remove it completly and check the mqtt integration in hassio.

wladimir79, 2021-03-21:
I have other device configured without autodiscover and I do not have any problem.
The entity in hassio is created and in the mosquitto logs is connected correctly

the only problem is that the value is not read, how can I be sure that it is sent by the Wemos?

mason-xx, 2021-03-21:
did you check if mqtt is recieving it? just subscribe to the topic and look how the payload looks

wladimir79, 2021-03-21:
i try with topic in the manual (wemos/wemosEM-XXXXXX/power), you see my config in the first post

Check the subscribe from hassio and from chrome addon but not received data 😔

mason-xx, 2021-03-21:
maybe check with just some toollike  https://mqttfx.jensd.de/ this show you your whole traffic on the server...

alcar21, 2021-03-21:
Hello, Sorry, i have no time for this project. I want to help you.

You can use MQTT Explorer too, it's a good program for explorer your MQTT topics.

Your config is wrong. Test the next config:
```
- platform: mqtt
  state_topic: "wemos/wemosEM-4E7EF6/power"
  name: "EM General A"
  icon: mdi:current-ac
  unit_of_measurement: "A"
  value_template: "{{ value_json.current }}"
- platform: mqtt
  state_topic: "wemos/wemosEM-4E7EF6/power"
  name: "EM General W"
  icon: mdi:power-plug
  unit_of_measurement: "W"
  value_template: "{{ value_json.watios }}"
- platform: mqtt
  state_topic: "wemos/wemosEM-4E7EF6/power"
  name: "EM General KWh"
  icon: mdi:power-plug
  unit_of_measurement: "KWh"
  value_template: "{{ value_json.kwh }}"
```
The config of home assistant is case sensitive.

wladimir79, 2021-03-21:
thank you so much, I didn't know about case sensitive

now everything works correctly :-) 

> Hello, Sorry, i have no time for this project. I want to help you.
>
> You can use MQTT Explorer too, it's a good program for explorer your MQTT topics.
>
> Your config is wrong. Test the next config: [...]

closed 2021-03-21 by wladimir79