alcar21/WemosEM #24: WiFi.psk

jimmyeao, 2021-08-16:
Downloaded latest release, infinito&masalla not accepted as wifi password, not sure where this is set in the code?

mason-xx, 2021-08-16:
its here https://github.com/alcar21/WemosEM/search?q=infinito

#define DEFAULT_SYSTEM_USER "wemosem"
#define DEFAULT_SYSTEM_PASSWORD "infinito&masalla"

should work

jimmyeao, 2021-08-16:
Got in thanks, was taking some time to start up it seems. Now has a problem connecting to MQTT
`Error in MQTT saved parameters: {"mqtt_username":"Error: mqtt username parameter incorrect","mqtt_password":"Error: mqtt password parameter incorrect"}`

Both are blank in my settings as I do not use this functionality in MQTT

jimmyeao, 2021-08-16:
OK fixed that, just entered "null" for both fields.

closed 2021-08-16 by jimmyeao